xuankuzcr/LIV_handhold #10: 请问这套设备的硬同步适用LIVOX_Horizon雷达与海康相机吗？

liunao123, 2024-02-18:
您好，我现在将雷达换成LIVOX_Horizon，相机还是海康，测试发现二者的时间戳是变成一样的了，但是底层对于雷达时间的改写一直不对劲，测试发现：雷达的sdk接受到了GPRMC消息，最后调用底层的LidarSetUtcSyncTime接口，这个接口虽然返回了0（也就是成功），但是其雷达的时间戳依然是从上电时刻起算的，GPRMC内的时间戳并没有赋值过去，请问您有碰到这种情况吗？您感觉有什么可能的原因呢？热切期待你的回复，谢谢 >_<

closed 2024-03-11 by liunao123

85256638, 2024-07-26:
大佬怎么结局了啊？希望关闭问题之前，把答案也分享下来。

xuankuzcr, 2024-07-26:
在我们之前[Camvox](https://github.com/ISEE-Technology/CamVox)的工作里，这个同步方案也是没问题的。建议重新pull一下最新的修复bug后的stm32固件和LiDAR driver，再看看是否能读取到GPRMC的时间戳。

liunao123, 2024-07-26:
@xuankuzcr 谢谢您的回复，我测试发现之前串口发送的GPRMC没有严格与官方给的一致，导致异常。更改其格式就正常了。
现在还有一个问题，不知道什么原因：GPRMC信息，串口单独去读，基本正常，但是livox的sdk去解析GPRMC信息的时候，会偶尔出现类似堵塞的情况，需要解析的内容会包含很多个GPRMC信息，最新的一帧是在最前面，所以我加了一个只读取最前面的GPRMC信息的一个逻辑，也就正常使用了。
您说的更新固件我找时间去试一下。有结果我反馈到这，谢谢您（握手). 
@85256638 您看一下我这个，看能不能解决您的问题。