xuankuzcr/LIV_handhold #16: 手持设备硬件设计方案

MisEty, 2024-03-19:
您好，感谢您开源如此优秀的工作来推动社区发展！现在我希望能搭建类似的LIV手持扫描设备复现FAST-LIVO，但对于硬件组成仍然有一些疑问：
1. 硬件系统的具体工作流程是怎样的？我的个人理解是STM32给雷达和相机分别发送同步的脉冲信号，雷达和相机接收信号后以同步的频率进行扫描，将结果传输给计算机执行后续SLAM任务
2. 设备中的on-board PC是必要的吗？是不是所有计算都在DJI manifold-2c上执行？能否使用笔记本电脑等算力更强的设备替代？
3. 是否一定要集成ros系统？我注意到所有的驱动都是集成到ros中的，FAST-LIVO也是以ros包的形式发布的，能否类似ElasticFusion那样作为不依赖ros的视线？
4. STM32在这个系统中是必要的吗，如果只是控制脉冲同步的话能否采用更简单的板子？
5. 如果使用其他的硬件，比如其他的相机型号，能否继续采用这个设计方案？

MisEty, 2024-04-16:
> 1.这个需要看代码去理解，具体原理也可以参考FAST-LIVO。【[https://arxiv.org/abs/2203.00893】](https://arxiv.org/abs/2203.00893%E3%80%91) 2.可以用笔记本替代mainfold，最开始我也是拿笔记本测试，但如果机械结构没固定好，很其实很容易出现数据丢包。 3.目前是依赖Ubuntu 和ROS的。 4.5 .MCU、相机是可以替换的。这里，只是提供一个验证过的方案，方便大家复现FAST-LIVO。

感谢您的方案，请问是否考虑开源STM32模拟输出GPRMC信号用于时间同步的代码，否则还依赖于外部的GPS设备提供GPRMC报文

xuankuzcr, 2024-05-27:
是pps+gprmc。LiDAR内部会进行10分频，以10Hz的频率进行分帧。

closed 2024-05-27 by xuankuzcr