xuankuzcr/LIV_handhold #19: Cabling configuration using livox convertor and avia

gc625-kodifly, 2024-06-05:
in the readme you said that PIN 11 is connected to RS485_B and PIN 12 is connected to RS485_A. I am using the livox convertor with the sync cable, so I just want to confirm whether RS485_A corresponds to pin 2 on the sync cable and RS485_B corresponds to pin 1 on the sync cable
[image]

xuankuzcr, 2024-06-28:
You can refer to these cabling configuration pictures.
If you use the converter2.0, follow this:
[image]
If you connect LiDAR directly to the STM32, follow this:
[image]
In our setup, LiDAR is directly connected, but both options are possible. Using a converter is more convenient.

xuankuzcr, 2024-06-28:
If you are using a converter, you can directly connect the TTL level generated by the STM32 to the sync cable. The RS485 conversion mentioned in the readme is not needed.

eung-ark, 2024-11-08:
hello. I am a student studying. First of all, thank you for sharing such great code and content.
 I have a question.
When connecting using converter 2.0 as in the case above, is the Ethernet port not connected to anything? Or should I connect the converter and PC with an RJ45 cable?

thank you

xuankuzcr, 2024-11-12:
> hello. I am a student studying. First of all, thank you for sharing such great code and content. I have a question. When connecting using converter 2.0 as in the case above, is the Ethernet port not connected to anything? Or should I connect the converter and PC with an RJ45 cable?
> 
> thank you

Yes, connect the converter to your computer with an RJ45 cable.