xuankuzcr/LIV_handhold #25: Is the lidar triggered like a camera?

SanghyunPark01, 2024-08-12:
The camera is a method of capturing an image when it is a rising edge by triggering at pps, when I look at the Livox time synchronization documentation, it seems to be a "time" (clock) synchronization operation, not a capture method through triggering. Does the lidar also have caputre data in the pps signal?

xuankuzcr, 2024-08-12:
Thank you for your attention. Livox LiDAR cannot capture data through a PPS signal trigger. The scan segmentation mechanism of Livox LiDAR is based on the integer seconds of ROS time. However, if you use PPS along with GPRMC (to create a simulated integer-second timestamp), it can effectively be considered as segmenting scans at the rising edge of the PPS signal.

SanghyunPark01, 2024-08-12:
Thank you for your answer. So in the case of the synchronization you mentioned here,
Does it mean that the camera image frame and lidar frame are **not** coming in at the same time, but "clock" is set to the GPRMCdata time?

xuankuzcr, 2024-08-13:
No, they can be considered as **being captured at the same moment**, with both the camera image frame and the LiDAR scan synchronized to the synthetic GPRMC time.

SanghyunPark01, 2024-08-13:
Thank you for your answer. I used flir machine vision camera and configured it similar to this repo (1Hz gPRMC, 30Hz pwm to cam). In rosbag record, one out of three times should be captured at the same time as LiDAR, but the time stamp of lidar and camera was not right.
(The camera captures the image at the rising edge.)

could you give me some advice on how to solve it?

xuankuzcr, 2024-08-13:
Please note that our camera driver has been modified. We store the timestamp from the LiDAR scan header into the shared memory and assign it to the corresponding camera frame. Your FLIR camera driver will also need to perform this operation. For every three image frames, copy the LiDAR scan timestamp to the camera frame. For the other two image frames, add 33.33ms and 66.66ms to the timestamp of the first frame, respectively.

SanghyunPark01, 2024-08-13:
Thank you so much for the advice! :smiley:

xuankuzcr, 2024-08-14:
You're welcome! If there are no further concerns, you can go ahead and close this issue.

closed 2024-08-14 by SanghyunPark01