xuankuzcr/LIV_handhold #27: 您好 打扰您了，我采用硬件设备为MID360+STM32做硬件同步，且使用串口助手检测PA9串口可以发出以下文本

wangdebin00223, 2024-09-07:
$GPRMC,000601.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2E
$GPRMC,000602.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2D
$GPRMC,000603.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2C
$GPRMC,000604.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2B
$GPRMC,000605.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A2A
$GPRMC,000606.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A29
$GPRMC,000607.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A28
$GPRMC,000608.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A27
$GPRMC,000609.00,A,2237.496474,N,11356.089515,E,0.0,225.5,080924,2.3,W,A*26
但是我将STM32的PA9和PB5分别接入MID360线的10号和8号线，在LivoxViewer查看Time Sync Type :显示 No Sync
并且有警告。能麻烦您帮忙分析一下，是哪里的问题吗？卡在这里好久了，一直排查不出来十分感谢!!!!
[image]
[image]
[image]

yunliuh, 2024-09-11:
接上地线试试

xuankuzcr, 2024-09-20:
可以在fast-livo仓库继续讨论，这里我有时候看不到，我只watch了livo的仓库。后续有问题在那里同步。

closed 2024-11-10 by wangdebin00223