xuankuzcr/LIV_handhold #38: Inquiry About FOV Calculation and Camera Selection

HyoyeonLee, 2025-02-07:
Hello,

I appreciate the work put into LIV_handhold, FAST-LIVO, and FAST-LIVO2, as well as the effort in making the code and datasets available.

I am currently in the process of selecting sensors and other components to set up the hardware for reproducing your work and have two questions.

**[Q1] FOV Calculation**

   According to the FAST-LIVO2 paper and [FAST-LIVO2-issue#34](https://github.com/hku-mars/FAST-LIVO2/issues/34), the Private Dataset uses the [MVL-HF0628M-6MPE](https://www.innosmart.bg/wp-content/uploads/CSM/HIKROBOT/product_docs/MVL-HF0628M-6MPE_datasheet_20221022--innosmart.bg--.pdf) (or [LM5JCM](https://pyramidimaging.com/specs/Kowa/LM5JCM.pdf)) lens with the [MV-CA013-21UC](https://www.rmaelectronics.com/content/HikRobot/CameraDatasheets/USB3.0AreaScan/MV-CA013-21UMUCDat%20Sheet.PDF?srsltid=AfmBOoo8hX5QWsF5TAEDIKFcoq8OJ77BMs2_9vRZ9F66LkQNOvIN79Lc) (1/2” sensor) camera, achieving a FOV of 70.6°(H) x 68.5°(V).
   However, based on the lens specifications of MVL-HF0628M-6MPE, the expected FOV for a 1/1.8” sensor is 63.11°(H) x 44.59°(V). 
   Given that a 1/2” sensor is smaller, I would expect it to result in an even narrower FOV of 54.8°(H) x 42.5°(V). 
   In the case of the LM5JCM lens, a FOV of 65.2°(H) × 51.3°(V) is specified for a 1/2" sensor.
   I would appreciate it if you could share how the FOV (70.6° × 68.5°) was determined.

**[Q2] Camera Replacement**

   Since the [MV-CA013-21UC](https://www.maxxvision.com/downloads/Cameras/USB30/Hikrobot/Hikrobot_MV-CU013-A0UMUC.pdf) has been discontinued, I am considering using  the MV-CU013-A0UC camera, as mentioned in [LIV_handhold-issue#24](https://github.com/xuankuzcr/LIV_handhold/issues/24). With the addition of BayerGB support to `mvs-ros-pkg`, would this camera be fully compatible?

   I appreciate your time and look forward to your insights.

Best regards,

xuankuzcr, 2025-02-16:
A1: The lens in the paper is custom-made, so it's not possible to find an exact match. The LM5JCM has a FoV of (Hor. x Ver. x Dia.) 82.4° x 66.9° x 95.0° on a 2/3" sensor. You could consider switching to a 2/3" CMOS sensor, such as the MV-CS050-10UC.

A2: Don't worry, it's fully compatible. If there are any issues, feel free to raise an issue, and I will respond promptly.

HyoyeonLee, 2025-02-17:
> A1: The lens in the paper is custom-made, so it's not possible to find an exact match. The LM5JCM has a FoV of (Hor. x Ver. x Dia.) 82.4° x 66.9° x 95.0° on a 2/3" sensor. You could consider [...]

Thank you so much for your help and confirmation! I was looking for a camera with an FOV similar to or slightly larger than my LiDAR's, so I’ve decided to go with the camera and lens you recommended. I really appreciate you sharing the model names, as it makes integrating them with your repo much easier.

xuankuzcr, 2025-02-17:
You're welcome! Glad I could help, and I hope the setup works perfectly for your project. If you ever need further assistance, don't hesitate to reach out. It's always great to see the community putting our work to good use—looking forward to seeing what you create! 😊

closed 2025-02-17 by HyoyeonLee