xuankuzcr/LIV_handhold #43: Is it possible to use a raspberry pi camera module v2?

GeomPozzoli, 2025-02-20:
Good morning,
congratulations for this open-source project
I wanted to try using a raspberry pi 4 and a livox mid-360 for data acquisition (data to be used later on another PC), but it is not clear to me if it is possible to use a pi camera module v2 as a camera, also considering the fact that it only has the ribbon cable and it is not possible to connect the opto-in and ground as in the project to the stm32.
Is it possible to manage data synchronization in another way, or do you have any suggestions for entry-level camera models compatible with the project's connections to do some experiments with?
Sorry if the question is a bit stupid but I don't have much knowledge about mvs cameras
Greetings
Stefano

xuankuzcr, 2025-02-21:
Hi Stefano,

Thanks for your question! For the Pi Camera Module V2, please check its manual to confirm if it supports external sync triggering. If not, you might need a camera that supports hardware triggering, such as FLIR Blackfly S, Basler ace series.

GeomPozzoli, 2025-03-05:
thanks for the reply,
I verified that the pi camera module v2 does not support external trigger hardware
I found a model of the inno maker CAM-IMX296Color-GS, which supports the trigger but the output format is YUV and not RGB, and I can't understand if it is compatible with the mvs_ros, do you have any info about it because in the code I see that there is only

> PixelFormat: 0 # 0: RGB8, 1: BayerRG8, 2: BayerRG12Packed, 3: BayerGB12Packed

Regards
Stefano

GeomPozzoli, 2025-03-06:
I probably need to use the camera type/brand related driver and not the mvs_ros_pkg.......................