xuankuzcr/LIV_handhold #45: Also imu and lidar time loop back

zhSlamer, 2025-02-23:
作者以及同学们你们好！我是遇到了激光雷达时间戳和IMU时间戳同时回退的问题
激光雷达为设备：Horizon
我认真审阅了issue32-imu loop  back的内容 #32，
一是供地：我将激光雷达的供电源的地与STM32地连接
二是我烧录的是最新的STM32源码，查看打印的GPRMC的时间符合hhmmss格式，符合满60进位,并且具有校验位
但仍然出现了时间戳周期性的回退，大致是4s内就会有变化，具体现象如下图：
向大家请求帮助，感谢作者和同学们！

[image]

[image]

[image]

zhSlamer, 2025-02-24:
观察数据，回退的时间为1s，即时间戳为68.9时，下一帧应为69.0，而此时又回到了68.0，即重复了时间戳，导致时间回退，并且经过几秒后，跳跃一秒，如从此时68.0开始68.0，68.1，...,**68.9，68.0，68.1**，68.2，...，72.8，**72.9，74.0，**74.1

[image]

[image]

zhSlamer, 2025-02-24:
问题解决了，因为TTL2USB丢包了，数据总是发布完整，还好我们有个新的TTL2USB，换了之后立马好了

[image]

closed 2025-02-25 by zhSlamer

wwtx, 2025-05-07:
@zhSlamer 请问如何查看TTL2USB是否丢包了？

zhSlamer, 2025-05-08:
> [@zhSlamer](https://github.com/zhSlamer) 请问如何查看TTL2USB是否丢包了？

你使用cutecom看看数据每一次打印的是否是完整的一整条数据，如果还不行，就改波特率