xuankuzcr/LIV_handhold #46: docker下timeshare的读写问题

LauZanMo, 2025-02-24:
你好，感谢你的开源！

我现在正在尝试将这个工程集成进docker中，遇到了以下问题：

因为docker中默认使用的root权限，我将path_for_time_stamp的构建从：

```c++
  const char *user_name = getlogin(); // root权限下为nullptr
  std::string path_for_time_stamp = "/home/" + std::string(user_name) + "/timeshare";
```

改为了

```c++
  std::string path_for_time_stamp = "/root/timeshare";
```

但是启动时会出现文件存在，mvs_ros_pkg读取时却为pointt为MAP_FAILED的问题，想问问您的看法。

如果需要，我可以提供docker镜像，再次感谢！

xuankuzcr, 2025-02-25:
你好，LiDAR的driver里面的路径也要改

LauZanMo, 2025-02-26:
上述问题的背景是我对mvs_ros_pkg和livox_ros_driver2的路径都做了修改，为了让我的问题更清晰，我已将docker镜像打包上传至dockerhub，执行：

```bash
docker pull lauzanmo/driver:latest
```

即可拉取镜像，执行：

```bash
docker run -it --rm --gpus all --net=host \
	-e NVIDIA_DRIVER_CAPABILITIES=all \
	-e DISPLAY=$DISPLAY \
	-v $HOME/.Xauthority:/root/.Xauthority \
	-e LD_PRELOAD=/opt/MVS/lib/64/libMvCameraControl.so \
	-e MVCAM_COMMON_RUNENV=/opt/MVS/lib \
	--device /dev/bus/usb/ \
	-v /dev/bus/usb/:/dev/bus/usb/ \
	-v /opt/MVS/:/opt/MVS/ \
	driver:latest \
	bash -it -c "roslaunch livox_ros_driver2 handhold.launch"
```

即可在容器中运行，handhold.launch即为同时启动livox_ros_driver2 rviz_MID360.launch和mvs_ros_pkg mvs_camera_trigger.launch。

构建镜像的dockerfile如下：

```dockerfile
ARG BASE_IMAGE=ros:noetic
ARG ROS_WS=/root/catkin_ws
FROM ${BASE_IMAGE}

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    git wget nano libomp-dev build-essential python3-pip libopencv-dev \
    ros-${ROS_DISTRO}-eigen-conversions ros-${ROS_DISTRO}-cv-bridge \
    ros-${ROS_DISTRO}-image-transport-plugins \
    ros-${ROS_DISTRO}-pcl-ros ros-${ROS_DISTRO}-rviz && \
    apt-get clean && rm -rf /var/lib/apt/lists/*

WORKDIR /root
RUN git clone https://github.com/Livox-SDK/Livox-SDK2.git && \
    mkdir -p Livox-SDK2/build && \
    cd Livox-SDK2/build && \
    cmake .. && \
    make -j$(nproc) && \
    make install && \
    rm -rf /root/Livox-SDK2

COPY ./MVS/ /opt/MVS/

ARG ROS_WS
ENV ROS_WS=${ROS_WS}
COPY LIV_handhold/livox_ros_driver2 ${ROS_WS}/src/livox_ros_driver2
COPY LIV_handhold/mvs_ros_pkg ${ROS_WS}/src/mvs_ros_pkg
ENV LD_PRELOAD=/opt/MVS/lib/64/libMvCameraControl.so
WORKDIR ${ROS_WS}/src
RUN cd livox_ros_driver2 && \
    /bin/bash -c '. /opt/ros/${ROS_DISTRO}/setup.bash; ./build.sh ROS1' && \
    rm -r ${ROS_WS}/build/ && \
    sed --in-place --expression '$isource \"${ROS_WS}/devel/setup.bash\"' /ros_entrypoint.sh

WORKDIR /root
ENTRYPOINT ["/ros_entrypoint.sh"]
CMD ["bash"]
```

希望能得到你的帮助！

LauZanMo, 2025-02-27:
刚刚做了@pcl5 建议的尝试，都修改为“/tmp/timeshare”，依然是文件被创建，但是无法读取。

[image]