xuankuzcr/LIV_handhold #50: launch运行错误

dawalishi122, 2025-03-16:
roslaunch hikrobot_camera hikrobot_camera.launch 
... logging to /home/nuc/.ros/log/044602c6-0262-11f0-94fa-fbcea1d3ebae/roslaunch-nuc-NUC13VYKi5-235393.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nuc-NUC13VYKi5:45095/

SUMMARY
========

PARAMETERS
 * /ExposureTime: 70000
 * /FrameRate: 10
 * /FrameRateEnable: True
 * /GainAuto: 2
 * /Gamma: 0.7
 * /GammaEnable: True
 * /Offset_x: 0
 * /Offset_y: 0
 * /Saturation: 128
 * /SaturationEnable: False
 * /TriggerMode: 1
 * /TriggerSource: 0
 * /height: 320
 * /rosdistro: noetic
 * /rosversion: 1.17.0
 * /width: 640

NODES
  /
    hikrobot_camera (hikrobot_camera/hikrobot_camera)

ROS_MASTER_URI=http://localhost:11311

process[hikrobot_camera-1]: started with pid [235422]
[device 0]:
UserDefinedName:

set AcquisitionFrameRateEnable OK! value=1.000000
set AcquisitionFrameRate OK! value=10.000000
set Height OK!
set Width OK!
set Offset X OK!
set Offset Y OK!
set ExposureTime OK! value=70000.000000
set GammaEnable OK! value=1.000000
set Gamma OK! value=0.700000
set GainAuto OK! value=2.000000
set BalanceRatio OK! value=0.000000
set TriggerMode OK!
set PixelFormat OK ! value = RGB
PixelFormat :17301514!
[ INFO] [1742131044.363636623]: The Number of Faild Reading Exceed The Set Value!

[hikrobot_camera-1] process has died [pid 235422, exit code 255, cmd /home/nuc/catkin_ws/devel/lib/hikrobot_camera/hikrobot_camera __name:=hikrobot_camera __log:=/home/nuc/.ros/log/044602c6-0262-11f0-94fa-fbcea1d3ebae/hikrobot_camera-1.log].
log file: /home/nuc/.ros/log/044602c6-0262-11f0-94fa-fbcea1d3ebae/hikrobot_camera-1*.log

参考了其他的教程，测试修改line为其他的（0,1,2）均不可行，修改timeshare权限也不行。

Lianghongnan, 2025-03-23:
请问下你的相机型号是什么呢？

dawalishi122, 2025-03-23:
我解决了，就是港科大开源资料中提到的海康彩色相机。

ฅ՞•ﻌ•՞ต

Lianghongnan, 2025-03-23:
额，是港大[@xuankuzcr](https://github.com/xuankuzcr)开源的

dawalishi122, 2025-03-23:
感谢纠正!

ฅ՞•ﻌ•՞ต