xuankuzcr/LIV_handhold #51: error when running mvs_camera_trigger.launch

jml-creator, 2025-03-18:
Thank you for sharing your code and expertise.

I am encountering an issue when trying to run `roslaunch mvs_ros_pkg mvs_camera_trigger.launch`. The following error is displayed:

`class_loader::LibraryUnloadException`

Could you kindly point me in the right direction to resolve this?

# parameters

1. hikrobot camera driver works fine

2. running from docker container 

3. no error during build

4. rosdistro: noetic

5. opencv 4.2.0

## error message

root@pl-NUC13ANKi5:/ros_ws# roslaunch mvs_ros_pkg mvs_camera_trigger.launch
... logging to /root/.ros/log/56f9fe00-031b-11f0-83df-e0c2643980a5/roslaunch-pl-NUC13ANKi5-1749.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pl-NUC13ANKi5:37545/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    mvs_camera_trigger (mvs_ros_pkg/grabImgWithTrigger)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [1757]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 56f9fe00-031b-11f0-83df-e0c2643980a5
process[rosout-1]: started with pid [1767]
started core service [/rosout]
process[mvs_camera_trigger-2]: started with pid [1770]
process[rviz-3]: started with pid [1775]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
MESA: warning: Driver does not support the 0xa7a0 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa7a0 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
[ INFO] [1742207641.325130219]: Successfully created handle for device 0
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
[mvs_camera_trigger-2] process has died [pid 1882, exit code -6, cmd /ros_ws/devel/lib/mvs_ros_pkg/grabImgWithTrigger /ros_ws/src/mvs_ros_pkg/config/left_camera_trigger.yaml __name:=mvs_camera_trigger __log:=/root/.ros/log/56f9fe00-031b-11f0-83df-e0c2643980a5/mvs_camera_trigger-2.log].
log file: /root/.ros/log/56f9fe00-031b-11f0-83df-e0c2643980a5/mvs_camera_trigger-2*.log
[mvs_camera_trigger-2] restarting process
process[mvs_camera_trigger-2]: started with pid [1905]
[ INFO] [1742207645.394004647]: Successfully created handle for device 0
^C[rviz-3] killing on exit
[mvs_camera_trigger-2] killing on exit

Harland666, 2025-03-27:
I also encountered the same problem. May I ask how you solved it?

jml-creator, 2025-03-28:
Harland666, I have not solve the issue yet. Looking forward to fix this.

Harland666, 2025-04-02:
Just make sure the parameters in MVS are set the same as those in the configuration file.[image]

Harland666, 2025-04-02:
Just make sure the parameters in MVS are set the same as those in the configuration file.

smallCuZ, 2025-04-03:
> Just make sure the parameters in MVS are set the same as those in the configuration file.

Thank you, the problem has been resolved！

closed 2025-04-04 by jml-creator