xuankuzcr/LIV_handhold #56: 运行mvs_camera_trigger.launch 无法捕获相机的图像

Victorrr1, 2025-04-14:
复现的时候使用的是海康MV-CU013-A0UC工业相机：
yaml文件配置如下：
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
TopicName: "left_camera/image"

TriggerEnable: 0 # 0 stands for Off, 1 stands for On

ExposureAutoMode: 0 # 0 stands for Off, 1 stands for Once, 2 stands for Continues
ExposureTime: 80000 # us

# ExposureAutoMode: 2
# AutoExposureTimeLower: 100
# AutoExposureTimeUpper: 20000

image_scale: 1 # 1 0.5
GainAuto: 2 # Gain Auto, 0 stands for Off, 1 stands for Once, 2 stands for Continues
Gain: 15 # min: 0   max: 17.0166
Gamma: 0.7  # min: 0   max: 17.0166
GammaSelector: 1 # 0 stands for user, 1 stands for sRGB
# GammaEnable: 1

PixelFormat: 3 # 0: RGB8, 1: BayerRG8, 2: BayerRG12Packed, 3: BayerGB12Packed

运行的时候：
MV_CC_ConvertPixelType failed! nRet [80000000], skipping this frame
^CSrcPixelType: 0x10c002c, Width: 1280, Height: 1024
SrcDataLen: 1966080, DstBufferSize: 1966080

海康SDK转换像素类型是失败，但是我查看了SDK的头文件输入类型和输出类型都是支持的 不知道是什么问题了。。

MiracleVae, 2025-04-14:
我也遇到了

xuankuzcr, 2025-04-14:
你俩相机型号都是多少

Victorrr1, 2025-04-14:
我用的是MV-CU013-A0UC

发自我的iPhone

MiracleVae, 2025-04-14:
我是CS060-10GC

xuankuzcr, 2025-04-14:
@Victorrr1 你相机支持 “Bayer GB 8/10/10Packed/12/12Packed“ 见 https://www.hikrobotics.com/cn/machinevision/productdetail/?id=12628
把PixelFormat设为3试试
https://github.com/xuankuzcr/LIV_handhold/blob/ae191786797f48d3672ab812fa896b8f06f17fd5/mvs_ros_driver/config/left_camera_trigger.yaml#L25

xuankuzcr, 2025-04-14:
@MiracleVae 你的相机支持BGR 8，见 https://www.hikrobotics.com/cn/machinevision/productdetail/?id=7482 具体运行报错是啥

xuankuzcr, 2025-04-14:
我明白了，上次添加多相机支持的时候有些地方没考虑Bayer格式，你把所有出现`stParam.nCurValue`的地方改成`stParam.nCurValue*3`试试@Victorrr1

Victorrr1, 2025-04-14:
这个我尝试了 好像不太行

不过我注意到我这个版本的海康MVS驱动MvCameraControl.h 中似乎没有MV_CC_ConvertPixelType的声明 里面有一个MV_CC_ConvertPixelTypeEx函数 我现在还在调试 不知道是不是这个的问题 如果我这边调试成功了 会更新这个issue

MiracleVae, 2025-04-14:
感谢您回复！
我的报错和那位相似，运行mvs_camera_trigger.launch时rviz中没有图像，终端不断报MV_CC_ConvertPixelType failed! nRet [80000000], skipping this frame.
配置文件中换了图像格式也是这样.

Victorrr1, 2025-04-14:
还是无法正常显示。。

MiracleVae, 2025-04-14:
按您的提示改了还是不行，报错没有变化，请问是否与MVS官方客户端的版本有关，我用官方客户端可以正常显示图像。

xuankuzcr, 2025-04-14:
要不加下我联系方式吧，Github主页有，一起看看问题 @MiracleVae @Victorrr1

AshadowZ, 2025-04-14:
我使用的是MV-CU013-A0UC，添加了BayerGB8的像素格式。然后按作者说的将所有出现stParam.nCurValue的地方改成stParam.nCurValue*3，就可以正常出图了。不然会报MV_E_BUFOVER的错误码，先缓存已满
[image]

[image]

Victorrr1, 2025-04-14:
好的@xuankuzcr  非常感谢郑博的指导

Victorrr1, 2025-04-14:
@AshadowZ 十分感谢您的回复！
根据您这个上面的错误码 我这个似乎是另一个问题 是错误或无效的句柄

MiracleVae, 2025-04-14:
@Victorrr1 你好，我这边可以看到图像了，你可以按我的方法试试。把MVS官方客户端降低版本，我原来是3.0.1，换成2.1.2，然后重新开一个终端source一下就可以了。这是2.1.2的链接https://download.csdn.net/download/qq_45445740/90146864?spm=1001.2014.3001.5501

Victorrr1, 2025-04-14:
好的 感谢您的分享 我马上尝试一下！

Victorrr1, 2025-04-14:
我来总结一下 出现MV_CC_ConvertPixelType failed 错误码80000000 需要把海康的驱动版本回退到2.1.2 这个时候运行代码会出现MV_CC_ConvertPixelType failed 错误码80000002 只需要按照

> 我明白了，上次添加多相机支持的时候有些地方没考虑Bayer格式，你把所有出现`stParam.nCurValue`的地方改成`stParam.nCurValue*3`试试，如果可以的话我更新上去，麻烦了 [@Victorrr1](https://github.com/Victorrr1)

将代码的buffer全部开成原来的三倍即可

Victorrr1, 2025-04-14:
非常感谢郑博的指导 也感谢 @MiracleVae @AshadowZ 两位老哥的指导 我马上把该issue关闭

closed 2025-04-14 by Victorrr1

Victorrr1, 2025-04-15:
@AshadowZ 你是对的 我昨天fast-livo2运行的时候发现libusb的动态库链接到海康的MVS中去了 修改了一下bashrc中动态库的顺序 发现运行相机驱动的时候又报错了，这次报了MV_CC_ConvertPixelType failed! nRet [80000006], skipping this frame（看样子它会随机报一个错？）

然后我重装了一下MVS，发现它会修改系统的bashrc并将自己lib的优先于系统的lib。 相机驱动能运行的关键是要用海康MVS的libusb库 然后fast-livo2运行的时候要隔离一下MVS的lib使用系统的libusb库即可。

quinnym233, 2025-07-29:
export LD_LIBRARY_PATH=/opt/MVS/lib/64:/opt/MVS/lib/32:/opt/MVS/lib/64:/opt/MVS/lib/32  指定库后还是报错[mvs_camera_trigger-1] restarting process
process[mvs_camera_trigger-1]: started with pid [15082]
/home/ym/fast_livo_ws/devel/lib/mvs_ros_driver/grabImgWithTrigger: error while loading shared libraries: libmessage_filters.so: cannot open shared object file: No such file or directory

xjy0410, 2025-09-01:
> export LD_LIBRARY_PATH=/opt/MVS/lib/64:/opt/MVS/lib/32:/opt/MVS/lib/64:/opt/MVS/lib/32 指定库后还是报错[mvs_camera_trigger-1] restarting process process[mvs_camera_trigger-1]: started with pid [15082] [...]

我也遇到了相同的问题，请问解决了吗？